davidrg/ckwin #1: Unable to transfer files to a device

GovindRulz, 2015-05-26:
I have trying to send file over to a linux based device , but k95 keeps crashing . tried running it as administrator . , attempted logging too . the log file gets generated but its empty . am kind of new to this , please help . thank you :)

GovindRulz, 2015-06-03:
:( :(

davidrg, 2015-06-03:
How are you trying to transfer the file?

The current codebase is missing SSH support (US cryptography export regulations prevented its release so it needs rewriting). To talk to a linux machine you'll need to use an RS232 cable or a telnet connection with C-Kermit running on the machine at the other end (or use FTP).

If you need to transfer files using SSH then WinSCP may be a better tool for now.

GovindRulz, 2015-06-04:
Thank you so much for your response David. Am connecting to a linux machine
through the RS232 Cable , I have c-kermit , running on the machine . I
compiled your source code , executed the exe passing parameters via a
config file using the 'take' command.

Im able to transfer files to and from the device using Hyper terminal , But
when i use K95 or the executable from your source , the executable either
crashes mid way , it says there is an error in the transfer.  I tired to
generate a log , the file is getting created but there is no data in it
.Please note my PC is windows 7 . I want to integrate this file transfer
capability with a dot net application , ive been successful in
communicating with device over the COM port , using my dot net app.

Listed below is the sequence of commands passed via command file

#set line /dev/ttyS0
set line /dev/ttyUSB0
#set line /dev/ttyACM0
set port COM9
set speed 115200
set carrier-watch off
set handshake none
set flow-control none
robust
set file type bin
set file name lit
set rec pack 1000
set send pack 1000
set window 5
set dial display on
set file collision overwrite
SET SESSION-LOG TEXT
log session log.txt
#kermit -r
#kermit -s c:\\Hello.txt

Govind
+91-9995822546

govind@horizontechsolutions.in

davidrg, 2015-10-22:
Looks like this might be a compatibility issue with the compiler used in the more recent test builds (Visual C++ 2010). Sometime in the next week I hope to get a bit of time to look into this in more detail and try to fix it. In the meantime if you're still looking to use C-kermit for windows, earlier test builds done with Visual C++ 6 are apparently able to transfer files successfully, the most recent of which is available here: http://ftp2.zx.net.nz/pub/Kermit95/test_builds/3_0-21-nov-2013/winkermit-3.0-test-21-nov-2013-vc6.zip

davidrg, 2022-07-22:
Should be fixed now - all versions of Visual C++ from 6.0 SP6 through to 2022 should be able to complete file transfers successfully now.

closed 2022-07-22 by davidrg