davidrg/ckwin #60: Change default paths and files

davidrg, 2022-08-25:
C-Kermit for Windows is descended from Kermit 95, but it is not Kermit 95. So it probably should be looking for its own files first and only falling back to Kermit 95 things for ease of migration and backwards compatibility.

More information on what Kermit 95 (and therefore CKW currently) looks for is documented here: https://kermitproject.org/k95readme.html#x9.1

- [ ] set \v(common) and \v(inidir) to C:\ProgramData\C-Kermit if it exists, else C:\ProgramData\Kermit 95 if it exists else \v(exedir)
- [ ] set \v(appdata) to %USERPROFILE%\AppData\Roaming\C-Kermit if it exists, else %USERPROFILE%\AppData\Roaming\Kermit 95
- [ ] set \v(download) to the standard downloads director (%USERPROFIEL%\Downloads ?)
- [ ] Don't auto-create \v(appdata) or \v(common) if they don't exist as they're not actually required and many of the subdirectories are probably obsolete.
- [ ] Look for ckw.ini before k95.ini
- [ ] rename default k95.ini to ckw.ini and adjust to search for ckwsite.ini before k95site.ini, and ckwcustom.ini before k95custom.ini

mbernardi1961, 2024-06-15:
Also all files currently named k95*.* should be renames to ckw*.* with appropriate changes in the code.
Update to mirror changes in C-Kermit 10 pre-beta 11

davidrg, 2024-08-12:
The decision has been made to switch back to the K95 branding (https://github.com/davidrg/ckwin/issues/287) and future versions will be a direct continuation of Kermit 95 so this no longer needs to be done.

closed 2024-08-12 by davidrg