davidrg/ckwin #93: Unable to send file with k95.exe on embedded target using ekermit

noomio, 2022-09-02:
Hi,

I'm able to use teraterm to send files to the target via kermit protocol (large files eve 36k)

But when I try to use this windows tool it fails.

I can't pin point what the reason might be and the debug log isnt that useful...

Serial is not using any hardware handshaking or flow control.

Any ideas? 

Small file is OK:
[good.txt](https://github.com/davidrg/ckwin/files/9476077/good.txt)

[bad.txt](https://github.com/davidrg/ckwin/files/9476108/bad.txt)

davidrg, 2022-09-02:
Are you able to test if C-Kermit on linux has the same issue communicating with your ekermit setup? That would help narrow down if its a problem with C-Kermit in general or with the Windows version of C-Kermit specifically.

I know Frank da Cruz, who maintains C-Kermit for other platforms, is investigating a bug logged by the Teraterm people about issues with long packets so its possible this might be the same problem.

Any details you can provide about the way you've got ekermit setup would be helpful too

noomio, 2022-09-03:
Worked like a charm on Linux.

Using VM and the same USB to serial convert.

```
stty -F /dev/ttyUSB0 115200  –cstopb –ixon -parity
gkermit -X -d ./debug.log -e 128 -T --x -p n -s test.txt < /dev/ttyUSB0  > /dev/ttyUSB0
```

[debug.log](https://github.com/davidrg/ckwin/files/9481583/debug.log)

davidrg, 2022-09-03:
I don't suppose you'd be able to give C-Kermit 10 a go? 10.0 beta 04 is what C-Kermit for Windows is currently based on - you'd probably have to compile it from source yourself unfortunately: https://www.kermitproject.org/ckdaily.html. If you get the same problem there then its a more general C-Kermit 10 problem, possibly the same one Frank is currently investigating.

If C-Kermit 10.0 beta 04 on Linux works fine then I guess its a Windows-specific issue and I'll have to find some way of reproducing it here. I know network file transfers between C-Kermit for Windows and the other Kermit implementations (C-Kermit 9.0.302 from the debian repos, G-Kermit CU-1.00, MSKermit) do work in general but its possible there is some compatibility issue with E-kermit or serial connections specifically.

noomio, 2022-09-03:
Hi,

It fails on Linux when I compiled it.

Looks like the same as windows.

[linux C-Kermit 10 beta 4.txt](https://github.com/davidrg/ckwin/files/9481731/linux.C-Kermit.10.beta.4.txt)

davidrg, 2022-09-03:
Awesome! That should help a lot. It being a problem with C-Kermit in general should make this easier to fix I hope. I'll forward all this on to Frank and hopefully this can be fixed in not too long.

In the meantime if you're after a C-Kermit for windows implementation that might work, this old build from 2013 should be based on C-Kermit 9.0 and might not have this bug:
https://ftp2.zx.net.nz/pub/Kermit95/test_builds/3_0-21-nov-2013/

noomio, 2022-09-04:
Thanks.

This .exe worked!

davidrg, 2022-09-05:
Frank da Cruz is now investigating this problem but he doesn't have a GitHub account. He is having trouble reproducing the problem - any chance you could flick him an email? He can be reached at fdc@kermitproject.org

noomio, 2022-09-08:
Ok, I shall send an email the old school way ;)

commit 51ffe263 (2022-09-15): Commit long-packet fix from FDC

This fixes #93 plus potentially issues transferring files to/from TeraTerm
closed 2022-09-15 by davidrg