davidrg/ckwin #150: Passing argument to k95

noomio, 2023-02-01:
Hi,

I pass arguments to k95 as below. However I get prompted:

? SET speed as no effect without prior SET LINE

This works on my PC and com port opens but on another PC its doesn't seem to open the com port. 
Whats the best way to pass these arguments?

```
				Start-Process -Wait -FilePath "./ckw-b4/k95.exe" -ArgumentList "-Y -C `"set modem type none, set line $com_port, set carrier-watch off, set speed 115200, set flow none, set parity none, set stop-bits 1, set window 1, set delay 1, OUTPUT \10, INPUT 1 \10>>, if success OUTPUT ekermit -s $File\10, INPUT 1 ekermit -s $File\10, if success receive, if failure OUTPUT \3, OUTPUT \3, OUTPUT \3, EXIT`""  ```

davidrg, 2023-02-01:
The command line arguments all look fine to me - I don't see any obvious reason why it would fail on some machines. Only thing I can think of is it doesn't like the value supplied for $com_port. 

Looking at the code, it looks like on Windows C-Kermit only understands the serial ports com1-com8 (you can also skip the 'com' bit and just give a number 1-8). A com port with a higher number may not work though what (if any) error message you'd get in that scenario I couldn't say.

If add on the -d argument it should produce a debug log (named debug.log) in the current directory which may provide further hints as to what's going wrong.

noomio, 2023-02-01:
Hi,

Actually it was a leftover from tera term that removed the COM part when passing the variable. All good now!

closed 2023-02-01 by noomio

davidrg, 2023-02-02:
Great to hear!