davidrg/ckwin #168: ckw-b4.zip flagged as virus

cleeus, 2023-03-14:
Hello David,

the [B4 release](https://github.com/davidrg/ckwin/tree/10.0beta.08-4) .zip is consistently flagged as a virus on multiple virus scanners. Virus scanner is complaining about iksdsvc.exe. Could you maybe tag a new build?

johnsonjh, 2023-03-14:
I’ve seen this before too myself - on even the simplest “Hello World” program built with some MSVC versions.  I see it’s been reported:

[image]

davidrg, 2023-03-15:
Yeah, its a false positive. I put a fresh Visual C++ 14.2 build straight from the github build servers through virustotal with similar (though not as many) results: https://www.virustotal.com/gui/file/856a048aff6fd14f817f082247bdbe4c11428015ac0667a7a390040de3cba60e

This is the main reason the builds for modern windows doesn't include ctl3dins.exe (aside from it not being necessary there) - it kept tripping up my own virus scanner whenever I downloaded a build from github.

Not sure what, if anything, can really be done about it. I assume in this case the virus scanners are just reacting to the capabilities C-Kermit has - making and receiving network connections, transferring files, running as a service, etc.

closed 2025-02-12 by davidrg