davidrg/ckwin #233: Confirm serial connections are working properly on Windows 11

davidrg, 2023-09-16:
I had some difficulty when trying to use a USB Serial device on Windows 11 recently. PuTTY would connect to it but CKW Beta.04 just reported an error.

Could be user error, or a windows 11 incompatibility, or perhaps there really is a bug - perhaps 64bit related if so.

mbernardi1961, 2024-02-21:
I have a similar issue, while it appears to be possible to select a null modem in the gui, this doesn't seem to pass back to the command line. There is no way to set line [device name], I am using a usb to serial cable, which Window Control Panel says is Com3:
In linux using the same version of C-Kermit (10 Beta 5), the same device can be set via set line /dev/ttyUSB0
I would guess Serial connection isn't a priority these days, but for connecting to retro equipment is essential,

mbernardi1961, 2024-03-03:
Further investigation has determined that the port assignment can be found in device manager is (COM3) and can be set in C-Kermit by
set line COM3
I have used this to successfully connect a Windows laptop to a Raspberry Pi 4 running linux and C-Kermit 10 beta 10

davidrg, 2024-03-04:
I think the USB-Serial device I was originally having trouble with may just be faulty (PuTTY couldn't use it either). Around that time one or two other people had raised issues with using USB-Serial devices on modern windows too though so figured a bit more testing was warranted. I did a bit of that a few days back with another less generic USB-Serial device and it worked fine so might just be a combination of user error or specific devices having issues with recent Windows releases.

The old GUI Dialer app does have a selection of issues but unfortunately they're all a bit tricky to solve the app is pretty firmly trapped in the 90s due to a few 3rd party things it depends on. I plan to just rewrite it from scratch eventually.

davidrg, 2025-02-12:
I've done a whole lot of testing lately transferring data via serial (using an FTDI USB/Serial widget) to a Turbo XT and a 286 without any problems, so I think this can be closed.

closed 2025-02-12 by davidrg