davidrg/ckwin #304: SSH Support on ARM32

davidrg, 2024-09-08:
SSH support was disabled on ARM32 builds due to linker errors in libssh 0.10.3:

> misc.c.obj : error LNK2019: unresolved external symbol __imp_GetUserNameA referenced in function ssh_get_local_username
> misc.c.obj : error LNK2019: unresolved external symbol __imp_SHGetSpecialFolderPathA referenced in function ssh_get_user_home_dir

It would be nice to resolve this. Perhaps newer versions of libssh don't have this issue, or maybe the problems are easily patched away.

wmjb, 2024-09-09:
both those symbols are from Advapi32.lib, just add the library. -lAdvapi32 etc library is included in windows sdks

davidrg, 2024-09-18:
Turns out CMake was leaving out a bunch of the default libraries when targeting ARM32 for some reason I probably don't care about. Adding `-DCMAKE_C_STANDARD_LIBRARIES="kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib"` to the cmake command line seems to have been enough to at least get it to build.

This artifact (valid for 7 days) at least has SSH support present - whether it actually works or not is for now unknown: https://github.com/davidrg/ckwin/actions/runs/10918637849/artifacts/1946797310

closed 2024-09-23 by davidrg