davidrg/ckwin #364: Get WolfSSL building with OpenWatcom

davidrg, 2024-10-13:


jmalak, 2024-10-31:
Here is cpuid C in-line assembly function

```
extern void cpuid( unsigned int *reg, unsigned int leaf, unsigned int sub );
#pragma aux cpuid =      \
    ".586"                \
    "cpuid"               \
    "mov  [esi+0],eax"    \
    "mov  [esi+4],ebx"    \
    "mov  [esi+8],ecx"    \
    "mov  [esi+12],edx"   \
    __parm [__esi] [__eax] [__ecx] __modify [__ebx __ecx __edx]
```

jmalak, 2025-02-14:
Latest version of WolfSSL is buildable by OW 1.9 and OW 2.0.
Linux and Windows version is OK and OS/2 version implementation is in development.

davidrg, 2025-02-15:
Thats really awesome! Once I've got some work on C-Kermit for linux/unix out of the way, I'll have a go at building Kermit 95 with WolfSSL instead of OpenSSL to provide telnet-ssl/https/ftps support.

And even if Kermit 95 turns out not be directly compatible with WolfSSL right now, it should make building an SSH backend based on libssh2 (#395) much more possible, perhaps making SSH on Windows 2000 and older, and eventually OS/2, possible!