davidrg/ckwin #382: COMMENT macro is defined in os2.h, it is danger to use in ckwin code

jmalak, 2024-10-27:
use of COMMENT macro in ckwin code is very danger because it is defined in os2.h header file.
It doesn't report any message if it is used incorrectly with os2.h.
only in code is this macro undefined after including os2.h, but not everywhere.

I think it can be source of some issues for OS/2.

I think it should be renamed to something like CKCOMMENT or replaced by 0 (transparent solution).

davidrg, 2024-10-27:
Yeah, its been an issue for as long as the OS/2 port has existed. Everywhere os2.h is included, it *should* be followed by an `#undef COMMENT`, though I noticed a few places where that wasn't the case (now fixed).

The issues on OS/2 have _so far_ all just been bugs in Open Watcom; there is a list of the bugs encountered in #369 - not sure at this stage if they're only in Open Watcom 1.9, or if they affect Open Watcom 2.0 as well

jmalak, 2024-10-27:
I don't thing it is OW bugs, it is rather implementation issues, by example calling conventions problems.
Generaly there is always two calling convention.
first for OS calls and second is user defined (default is Watcom registers calling convention, not cdecl).
If code is based on microsoft compilers then default calling convention is cdecl.
Also use of calling conventions as _System in ckwin code is wrong way.
Typically _System calling convention is override by macros _System than _System calling convention cannot be used until _System macro is undefined. I submit PR "fix issue with status_func which was unportable #386" which use properly macros CKDEVDLL and CKDEVAPI which define calling convention and don't mask _System calling convention which must be always available but it isn't. CKDEVDLL define calling convention of DLL entries and CKDEVAPI define calling convention of user functions in DLL (pointers) not exported from DLL. Both can be changed as needed, but don't affect _System calling convention which is used by system calls. Existing code destroy _System calling convention that any next system calls are incorrect.

davidrg, 2024-10-27:
> I don't thing it is OW bugs, it is rather ckwin code implementation issues, by example calling conventions problems.

yeah, the calling convention stuff isn't bugs Open Watcom, but earlier some actual compiler and/or runtime bugs (listed in #369) were encountered which mostly caused crashes in various places, in particular in the NetBIOS code which is a mix of 16bit and 32bit stuff. MichalN has found workarounds for most of them (including #368, he is just tidying up his solution for it at the moment), though aside from #368 I've no idea if they affect Open Watcom 2.0 or not.

jmalak, 2024-10-27:
OW 2.0 is backward compatible with OW 1.9 that OW1.9 code should be compiled by OW 2.0, but has fixed some OW 1.9 bugs
OW 2.0 has much better C99 support.