davidrg/ckwin #394: Question about future direction for 16-bit code and EMX extender

jmalak, 2024-10-31:
I would like to ask about future support for EMX extender and 16-bit code.
- If you plan to support 16-bit code on any platform
- if you plan to support EMX extender
in case of no plans then some macros and related code can be removed.
It is related to following macros
```
__32BIT__
__EMX__
```

davidrg, 2024-10-31:
I'd *like* to get it building on 16-bit OS/2 again (see ticket #161), though its for now been a relatively low priority due to there still being a bunch of stuff broken on 32-bit OS/2 (like #152 - Open Watcom builds of the dialer just crash), plus important stuff to finish off on Windows to get it back to where the last commercial release was.

I don't know anything about the EMX stuff - I don't think any of this code supported DOS (there was [MS-DOS Kermit](https://kermitproject.org/mskermit.html) for that), so I guess it must have been for building the OS/2 version with GCC. Unless the 32bit version of GCC for OS/2 still needs that stuff I can't imagine it being used again.

jmalak, 2024-10-31:
OK that both should be hold in code for now.