davidrg/ckwin #441: Open Watcom 1.9 builds are currently failing

davidrg, 2025-02-01:
The issue seems to be the nmake clone which gives one of the following errors:

- Error(E08): Invalid macro name ()
- Error(E07): Expecting a Form Qualifier

It does not appear to be a result of any change in the K95 repo - commits that previously built fine with Open Watcom 1.9 now fail. In GitHub Actions they were failing reliably, while rerunning the build locally a few times might eventually get past the error.

The issue was first noticed on GitHub Actions, with failures starting from around 1:15PM on 1 February 2025 (UTC+13). Timestamps on generated artifacts are all around 00:15 AM on 1 February 2005. All GitHub Actions builds leading up to 1PM local time (midnight in github actions time) were completing without any trouble, and I was working on unrelated CI changes (pushing generated HTML fines to github pages - no change to source code, makefiles, or the Open Watcom parts of the workflow)

nmake.exe from Open Watcom 2.0 does not seem to suffer from this issue. So as a temporary workaround until this problem can be resolved properly, the GitHub Actions workflow has been adjusted to fetch nmake.exe from the latest OW 2.0 snapshot and use that instead of nmake.exe from OW 1.9.

jmalak, 2025-02-14:
I sorry, but no problem with current build on my repository.
It looks like damaged Open Watcom cache, try to delete it to be recreated.

jmalak, 2025-02-14:
I checked it with cleaned cache and I reproduce the issue.
It is caused by lowered permissions for CI-build.
I submit PR no. 444 which should fix it.

davidrg, 2025-02-15:
Thanks for finding that! The way it just stopped working had me really puzzled.

closed 2025-02-15 by davidrg