davidrg/ckwin #473: Dynamically Redefined Character Set support

davidrg, 2025-06-17:
Aside from the VT100 confidence tests (which don't make any sense for a terminal emulator), this is the **only** VT220 feature not currently supported by K95. With this implemented, K95s VT220 emulation would be as comprehensive reasonably possible.

This feature may also not be _especially_ difficult either. I guess the implementaiton would probably look something like:
 * Implement DECDLD Device control string which will involve parsing Sixels, turning them into an HBITMAP or something and sticking them into one of the DRCS tables. This will really be the hard part.
 * Then when the soft character set is designated, instead of doing character set conversion we'd set a soft-character attribute of some kind. _This would work if only one soft character set is supported, but what about multiple soft character sets?_
 * At rendering time, if the soft character set attribute is set rather than rendering the character with a font, the characters bitmap would be looked up in the DRCS table, resized to fit the font size, and then painted in that character position.

Unanswered questions:
* If we take the attribute path, how do we handle multiple soft character sets? While the VT320 (and, I assume, the VT220) only has one DRCS table, the VT510 has three. So we would probably need at least two attribute bits, and there is currently only one free. Might need to increase the size of the char attribute (which may not actually increase memory use depending on structure padding)
* How do we handle soft characters in scrollback? Perhaps when a soft character set is redefined we go clear the attributes for anything in scrollback?

Reasons to do this:
* Completes VT220 emulation, and might also do the same for VT320 emulation
* Provides sixel handling code for future graphics terminal features
* Ability to play [Termtris](https://github.com/jtsiomb/termtris)

Priority:
Low - nothing much uses it. Target some release post-3.0.

davidrg, 2025-10-24:
Your games are a large part of my motivation for getting this feature done. Before I stumbled on them, this and the recent VT420 features (paging, macros) were really just boxes that needed checking before I could justify adding a VT420 emulation option with no obvious software that actually makes use of any of it.

j4james, 2026-04-16:
> And this weekend I'll probably dig out the VT220 and if it still works to see if it applies the dot stretching [described here](https://www.masswerk.at/nowgobang/2019/dec-crt-typography) to soft-fonts too.

I'd be curious to know that too. I'm mostly interested in whether it doubles the 8th bit to produce full width glyphs. Because the VT240 _doesn't_ do that (at least in MAME), so there's no way to avoid a gap between characters.

j4james, 2026-04-18:
Its seems that the decuslib.com site is down, but I found a mirror of the folder I was looking for here:
https://ftp.digiater.nl/openvms/decus/vax85c/levine/vt200/

There are a bunch of `.com` files in that directory, but they're not executables - they're just text files which you can `type` or `cat` from a terminal shell to download the font. `ascii.com` is a plain ascii character set, `back.com` is a reversed version of the ascii font, `cyrillic.com` is obviously a cyrillic character set, etc.

The one I'm most interested to see is `full.com`, which should replace the entire character set with solid blocks. On the MAME VT240 (and my Windows Terminal DRCS implementation), you get gaps between the blocks, but I'm hoping your VT220 will extend the 8th bit to produce a full cell block.

j4james, 2026-04-18:
> Nothing intersting in DA1:

That's a shame. Maybe it predates them defining official extensions for those features.

> ED-2 erases everything - screen and scrollback.

I wonder if that's not actually scrollback in the modern sense, but just a logical buffer that's larger than the visible viewport (i.e. the windowing extension). If that's the case, then `CUP 1;1` should be able to move the cursor to the top of the "scrollback", and `CUP 999;1` would move you to the bottom. And if `DECHCCM` is supported, that would determine whether the cursor movement also panned the visible area.

> And as a bonus, SU and SD also seem to work.

In theory then, you might also be able to alter the range that can be scrolled with `DECSLPP`, and query the viewport position in that scrollable area with `DECRQDE`. And if pages are supported, maybe `NP`, `PP` and the rest of the paging sequences could work too.

j4james, 2026-04-18:
> In Australia, DEC modified the standard VT220 to create a VT220-Z (VT220 + VT131/2 block mode) as a special for the New South Wales Department of Health.

Btw, the Australian DEC division also made a VT225, which I think had similar functionality (block mode and multiple pages). It could also do ANSI colors, which I thought was cool.

And there are two manuals for that available [online](https://www.manualslib.com/products/Digital-Equipment-Vt225-3983033.html), which you may want to look at. I'm assuming at least some of the documentation will be relevant for your VT220. Unfortunately there doesn't seem to be a copy of the programmer reference available anywhere.

davidrg, 2026-04-18:
And the VT520 doesn't extend the 8th bit - you get gaps. Which I guess is fair as the font is text not full-cell:

[image]

Unlike the vt320 fonts I have, these all display pretty well on the VT520

davidrg, 2026-04-18:
Poking at the VT220 a bit more, I'm not entirely sure what purpose the character matrix size actually serves - at least for that terminal specifically.

From what I can tell:

- Cells are 9 columns wide
- Column 9 is not directly addressable
- Each pixel is an OR of that pixel, and the pixel to its immediate left (10100 will render as 11110).
- Character matrix size has no effect on the glyphs you can define. A font with a character matrix size of 2 (5x10) will still accept and render glyphs 8 columns wide
- Glyphs are not centered in the character cell at all. Column 1 in the glyph is always column 1 in the character cell.

So its seems to effectively be full-cell behaviour, but with dot-stretching and the 9th column not addressable.

davidrg, 2026-04-18:
That's a much better test pattern than what I was doing! DHL is a big help.

[image]

Looks like there is a gap on the left of the cell - guess that's the missing 10th column I hadn't spotted? If so, then is the right most column of cell N being copied into the left-most column of cell N+1?

> And I forgot to mention that the `ascii.com` font from the decus site is actually defined as a 132-column font, but I'm assuming it works for both 80-column and 132-column on the VT220. It doesn't work on a VT240 though, because that consider a 7x10 font to be invalid for 132-column mode.

I was wondering why my VT520 didn't like that one but hadn't got as far as investigating why - that explains it. At the moment in K95 I'm not doing anything with 80/132 column setting as without a narrow font to go with DECCOLM there didn't seem to be much point in treating the narrow soft-fonts specially. Though I do need to check if the VT520 will store both 80 and 132 column fonts in a single font buffer - if that's the case then I need to make sure that works and that the 80-column variant is picked over the 132-column one for rendering.

j4james, 2026-04-19:
The more I think about this, the more I think it may be better not to emulate that overlap exactly. For a couple of reasons:

1. If you're using the DEC Special Graphics character set to output line drawing characters (or anything full width), and you follow that with a full width DRCS character, they're not going to line up correctly if the DRCS character is offset by 1. Unless you're planning to apply this offset to all characters?

2. How exactly do you deal with rendition attributes for the overlapped column? Does it take the attributes of the source cell, or the attributes of the cell it is overlapping? If I had to guess I'd expect the VT220 to use the attributes of the overlapped cell, but that's likely to get weird when colour attributes are involved.

I do like the idea of emulating the original hardware exactly, but I think this level of detail may be more effort than it's worth. If I were doing this, I'd be inclined to only offset the text glyphs, so they're centred, but leave the full width glyphs with no offset, so they fit within the cell.

davidrg, 2026-04-19:
Oh, and last night I had a look into the VT320. While I don't have one of those to test against, or even the manual (_really_ wish `EK-VT320-RM` was online somewhere), [this page](https://vt100.net/dec/vt320/soft_characters) says a pixel is "nearly three times as high as it is wide"; I can't do nearly three times but exactly three times looks a little more correct than treating them as square:

[image]

[image]

[image]

[image]

[image]

[image]

j4james, 2026-04-19:
The VT320 cell size is 15x12 (for 80-column mode), so wouldn't you just scale that to match whatever your native cell size is? For example, if your native cell size is 10x20, you would need to scale the width down by 66%, but scale the height up by 166%. The pixel aspect ratio of the original device is somewhat irrelevant - it's the relationship between your cell size and their cell size that matters.

But if you do need to know the exact pixel aspect ratio, I'd estimate it to be 2.5:1, assuming a VT320 cell has a aspect ratio of 2:1 (which I think is typical for most terminals). With square pixels, a cell width of 15 would require a cell height of 30, so if you've only got 12 pixels height, those pixels would need to be stretched by 30/12 (i.e. 2.5).

closed 2026-07-04 by davidrg

j4james, 2026-10-12:
> Any idea if changing colour via DECCARA is supported on the VT525

@davidrg FYI, I finally did some testing of this on an actually VT525, and it seems it _doesn't_ support colour attributes in `DECCARA`. If you try to apply a 3x or 4x attribute to an area, those values are ignored. And if you try to reset all attributes of an area with a 0 parameter, the colours are left unchanged, and only attributes like bold/reverse/etc. are reset.

Personally I think that's a mistake, given the spec text I quoted above, so I think it's reasonable to keep supporting that functionality. However, if you ever do a separate VT525 terminal type, you may want to consider dropping colour support from `DECCARA` to make it more strictly compatible.

davidrg, 2026-10-12:
Thanks for confirming that! I've adjusted it so that this will be excluded from the future VT525 emulation profile - it will just be an extension for the K95 terminal type as it still seems a handy thing to support alongside changing crossed-out, overline, etc, via DECCARA.